vedderb/vesc_tool #335: Dataloss when reopen and overwriting an existing .vescpkg file

Tomblarom, 2023-11-08:
I just finished modifying an existing VESC package and it discarded all the made changes!
Under "Output" --> Choose file.. --> Selected existing .vescpkg --> confirmed override --> Save Package.

Its reproduceable. I expected the package to be overwritten and saved, instead of discarding my changes.. This is frustrating..

vedderb, 2023-11-13:
It should save when selecting the file in the latest beta. Before it would just update the path box, but not save.

Tomblarom, 2023-11-14:
Yes it saves now, but unfortunately it's not as intended. 😅 I was rather expecting the folder-icon to select an output path. Instead it loads the `test.vescpkg` and overwrites the contents in my newly written description field.

[image]

vedderb, 2023-11-14:
I realized that it should not save the file actually, I removed that change again. The intention is that you can open a package you have been working on and continue working on it. So the first step would be to select the file you will be working with and then work on it and save the changes as you go. I will put some effort into that editor soon to make it a bit easier to use and understand.

vedderb, 2023-11-14:
Hope this makes it more clear for now: https://github.com/vedderb/vesc_tool/commit/820f5e9d0706d9fc3de0fe4b64ce76debeee0761

Tomblarom, 2023-11-15:
Perfect. This is more what a user would expect, when an overwrite is about to happen!
[image]

Beyond that wasn't finding any issues. Loading/saving works fine. Might give some feedback over here: https://github.com/vedderb/vesc_tool/issues/337

closed 2023-11-15 by Tomblarom