vedderb/vesc_tool #336: Servo-Out Firmware for Enertion FOCBOX

mgagvani, 2023-11-08:
I have an Enertion Focbox 1.2 which I am trying to update to the latest firmware using the VESC tool. A picture of the model I have is below:

[image]

However, on the VESC tool software I do not see this motor controller under the "Hardware Version" list. The closest looks to be the UNITY motor controller but that is not the same as the Focbox 1.2. 

Where can I find firmware, with servo-out enabled, for the Enertion Focbox 1.2?

jaykup26, 2023-11-12:
When connected to the ESC, the VESC Tool will only show the firmware that matches the hardware version that is currently on the ESC.  When not connected to an ESC it will display all firmware built into the tool.  You don't really "select" firmware from the tool, it will just update what's on the ESC to the latest version by matching the hardware version when you hit the update button.

You can see what firmware is loaded on the device by looking at the bottom left of the Firmware tab under "VESC Firmware, Hardware and UUID".  On the right where it says "Supported Firmwares" that's the version that VESC Tool will load.

The original Focbox is based on the 4.12 hardware, just with a different physical layout and DirectFETs.  It uses the regular "410" hardware version of the firmware.

If for whatever reason the wrong firmware is on that ESC (like "Unity") then use the custom file option and flash it using the file from the archive repository.  

https://github.com/vedderb/vesc_fw_archive/tree/main/6.02/410_o_411_o_412

Use the "VESC_Default.bin" version and not the others which are for different current shunt resistance values in other modified 4.12 controllers.

Note that you may need to update the bootloader if the firmware isn't "sticking" using the custom file tab.  Firmware -> Bootloader -> Update

Those servo wires sticking out of the Focbox are an input into the ESC, and that needs to be enabled in the tool under App Settings -> General -> APP to Use -> PPM (or PPM and UART if you want to continue talking to the device with a UART dongle).  Then go to the App Settings -> PPM to change the control type.

An easier way to set that up is just to use the input wizard on the "Welcome & Wizards" tab.

Finally the 1.2 hardware version (1.6 was the last) of the Focbox had CAN bus issues.  If I recall correctly, it was when CAN bus is connected with VCC and GND along with CAN-H/CAN-L and only one Focbox is powered up, or they are not powered up at the same time, it will permanently damage/short the CAN bus chip on one of the ESCs and potentially short the entire 3.3v rail.  So it's recommended to only use CAN-H/CAN-L/GND and/or power them up at the same time.

mgagvani, 2023-11-13:
Thank you so much for your help @jaykup26! I was asking around on Discord and forums but got no response, so thank you.

I will try uploading the firmware you linked soon. However, the tutorials I've been following for my use case say that a firmware with `servoout` is needed to enable the servo output off the PPM pins. Is this the same thing as what you have described with the app?

vedderb, 2023-11-14:
With the latest firmware versions you don't need a different firmware to enable servo out, there is a setting for it now:
[image]

There is also a controls-tab where you can test the servo output:
[image]

mgagvani, 2023-11-15:
> You can see what firmware is loaded on the device by looking at the bottom left of the Firmware tab under "VESC Firmware, Hardware and UUID". On the right where it says "Supported Firmwares" that's [...]

I tried these steps:

1. I uploaded the `VESC_Default.bin` firmware. It was successful. However, when I re-connected it I got the message of "too old firmware" again. Thus, I updated the bootloader since it seemed like nothing changed. However, after I updated the bootloader I was not able to connect to the VESC at all and it went unrecognized by the tool. I'm not entirely sure what to do at this point since it's not even recognized now but it was at least recognized as a "too old firmware" before...

mgagvani, 2023-11-15:
Edit: I got another, identical ESC. I tried uploading the custom firmware you linked, but when I read the firmware version it does not change from v2.18. I didn't update the bootloader this time but updating the firmware doesn't seem to be doing anything.

jaykup26, 2023-11-21:
When the firmware is that old you have to load 5.03 or lower first using the custom file tab.  Then it can be updated to 6+.  Some sort of issue with the bootloader which changed after 5.03

https://github.com/vedderb/bldc/issues/615

mgagvani, 2023-11-24:
Got it. So if I upload the `VESC_default.bin` from here first (https://github.com/vedderb/vesc_fw_archive/tree/main/5.03/410_o_411_o_412) and then I upload the same firmware from here (https://github.com/vedderb/vesc_fw_archive/tree/main/6.02/410_o_411_o_412) it'll successfully retain the new firmware?

jaykup26, 2023-11-24:
Upload 5.03 using the custom tab. The bootloader is not updated when loading firmware this way which should make it work.

Then either load 6.02 the normal way by using the VESC Tool firmware update, which will auromatically update the bootloader, or manually update the bootloader using the bootloader tab then manually load 6.02 using the custom file tab.

If all else fails, use a cheap stlink v2 and the stlink utility to manually load the 6.02 firmware (then update the bootloader in the vesc tool)

mgagvani, 2023-11-28:
Thank you so much! That did end up working. Hopefully this will come in handy for someone in the future.

closed 2023-11-28 by mgagvani