vedderb/vesc_tool #356: flatpak is outdated

unitypunk, 2024-03-04:
https://github.com/flathub/flathub/pull/2043

flatpack installs version 3.0 rather then 6.02 (or i guess 6.05 is coming soon?)

@albfan

albfan, 2024-03-04:
I still miss a tag or branches from releases. The best info I can find is this:

https://forum.esk8.news/t/precompiled-vesc-tool-archive/15920

That matches with existing branches:

  origin/release_3_01
  origin/release_6_00
  origin/release_6_02

so probably we can do a new release with branch   origin/release_6_02

about 6.05, the commit just says its beta

unitypunk, 2024-03-05:
I just wasn't sure how close 6.05 was to release. No point in updating to 6.02 if 05 comes out next week

albfan, 2024-03-05:
There are problems to build this on current flatpak runtime. That's the main thing to resolve.

Where is the info about 6.05 will be released next week?

MigAP, 2025-01-31:
The flatpak version is still outdated (version 3.0.0) .

albfan, 2025-01-31:
Will give another try with last versión released

albfan, 2025-01-31:
vesc_tool release version 6.05, but as there's some kind of warning:

https://pev.dev/t/vesc-6-05-warning-psas-read-this-before-upgrading/1933

I will release both 6.02 and 6.05 so people can choose what to use

albfan, 2025-02-01:
Finally release 6.05 is on flapak, this can be closed

MigAP, 2025-02-03:
Super, thanks !