vedderb/vesc_tool #369: Mac VESC Tool Crashes upon BLE Connect

gamer736413, 2024-06-22:
On Mac, I'm having a new issue I didn't have before. When I scan BLE and select a device to connect to, VESC tool crashes. I have tried this on multiple devices that previously I could connect to fine for the past 1-2 years. 

I did try to uninstall and reinstall the latest build, but no change. 

MacBook Air M1 / Sonoma 14.3 (23D56)

blundey, 2024-08-28:
Also having the same issue on a macbook pro m2. Latest mac tool crashes upon BLE connect. It shows the device fine.

yoodog, 2024-09-28:
Version 6.05 stable for MACOS intel version - MacBook Pro 2019 i9, Also crashes every time after you select your BT VESC as soon as you press "connect" button... Has been doing it even on 6.02 VESC tool

I have even installed it on COMPLETELY fresh, as new, MacBook Pro i9 2019 (another one with similar specs) - same behaviour.

Windows VESC tool, if I switch to windows 10, connects via BT no issue.

Looks like MAC version has some bug that escaped DEVs

yoodog, 2024-09-28:
Can we please have it addressed? iPhone app connect perfectly fine