vedderb/vesc_tool #370: VESC Tool crashes and data does not load from log file

AidanNowa, 2024-06-25:
[image]

When attempting to do log analysis on a log stored in the micro SD of the VESC Express, the file will load, and the x-axis of the graph will update, but no data will appear in the data tab. Clicking on the blank graph will then cause the application to crash. 

The "Stats" tab appears to work properly.

vedderb, 2024-08-02:
This should be fixed in the later 6.05 betas. Can you give it a try?

AidanNowa, 2024-08-05:
I tried to replicate it again on the 6.05 version and have been unable to get the bug again. It might have been an issue on my end as I was duplicating logs on my machine and changing the separator to view in my version of Excel. When intentionally uploading a modified file now VESC tool doesn't show data/shows it improperly (since it is now formatted differently) but doesn't crash. I am not sure if this was what caused the initial problem but it's all I could think of. Either way, the issue of crashing appears to be resolved.

pkonnoth, 2024-10-28:
I've been experiencing the same issue and haven't found a solution yet. Does anyone know if this might be specific to macOS, or could it be a firmware bug? Any insights would be appreciated!