vedderb/vesc_tool #401: Inquiry for the addition of a voltage offset / voltage measurement correction

theauthenticjay, 2025-02-15:
I'm running into too many vesc controllers with inaccurate voltage ratings, some off by 0.1v, some off by a lot more. A lot of our riding apps have this voltage adjustment, but not vesc tool where we rely on it to be accurate for tilt back thresholds (onewheel based devices). I know it's really the controller manufactures fault for the inaccurate readings, but that doesn't change the fact that many of our controllers have these defects.

What're the possibly of something happening with that?

Thanks for your time, VESC has really changed and made my life a lot more enjoyable and fun!

hippo4123, 2025-02-15:
perchance if this feature could be implemented I would be of so grateful to you my Lord Vedder, Creator of the almighty VESC.

contactsimonwilson, 2025-02-15:
This would be better done in firmware than VESC Tool IMO. Getting different readings when connecting to a controller from a different device would be too confusing.

Khaos-01, 2025-02-15:
My brothers in VESC, we are gathered here today to address the implementation of a feature that would ultimately take our stoke to new levels.  A idea that if we were to see it brought to reality, would enhance the quality of life on our electric ridables immensely.   Once previously thought impossible.....no simon let us....ugh you right my king